FPGAwars/icestudio #714: [BUG] Icestudio claims I need to at least Python 3.5, when I have 3.10.12 installed

vguttmann, 2024-01-19:
I'm using Icestudio 0.6.0 on Ubuntu 22.04, and can't install any toolchain because of this.

cavearr, 2024-01-19:
Please, install latest wip from https://downloads.icestudio.io and tell me if it works.

vguttmann, 2024-01-19:
Unfortunately, I don't get downloads when installing from there - just this: 
[image]

vguttmann, 2024-01-19:
When downloading from GitHub, the icestudio.log contains the following errors: 
```
=======================================================================================
 Icestudio session Fri Jan 19 2024 22:34:19 GMT+0100 (Central European Standard Time)
 Version: 0.11
=======================================================================================
Node information:
  * Node version: v20.1.0
  * lts: undefined
  * SourceURL: https://nodejs.org/download/release/v20.1.0/node-v20.1.0.tar.gz

NW information: 
  * NW version: 0.76.1
  * NW-flavor: sdk
  * Chromium: 113.0.5672.64

System information:
  * Architecture: x64
  * Platform: linux

Profile file: /home/vguttmann/.icestudio/profile.json

- PROFILE:

{
  "data": {
    "board": "upduino3",
    "boardRules": true,
    "collection": "/tmp/.mount_Yc1cqo/usr/bin/resources/collection",
    "language": "en",
    "uiTheme": "dark",
    "externalCollections": "",
    "externalPlugins": "",
    "remoteHostname": "",
    "showFPGAResources": false,
    "displayVersionInfoWindow": "no",
    "lastVersionReview": "0.6.0",
    "loggingEnabled": false,
    "loggingFile": "",
    "pythonEnv": {
      "python": "",
      "pip": ""
    }
  }
}

- PATHs

common.BASE_DIR: Icestudio base dir: /home/vguttmann
common.ICESTUDIO_DIR: Icestudio folder: /home/vguttmann/.icestudio
common.PROFILE_PATH: Profile path: /home/vguttmann/.icestudio/profile.json
common.INTERNAL_COLLECTIONS_DIR: Internal collections: /home/vguttmann/.icestudio/collections
common.APIO_HOME_DIR: APIO folder: /home/vguttmann/.icestudio/apio
common.ENV_DIR: Python virtual environment: /home/vguttmann/.icestudio/venv
common.ENV_BIN_DIR: Executable files: /home/vguttmann/.icestudio/venv/bin
common.ENV_PIP: PIP executable:  /home/vguttmann/.icestudio/venv/bin/pip3
common.APIO_CMD: Apio command: APIO_HOME_DIR="/home/vguttmann/.icestudio/apio" PATH="/home/vguttmann/.icestudio/venv/bin:/tmp/.mount_1fQc9L/usr/bin:/tmp/.mount_1fQc9L/usr/sbin:/home/vguttmann/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" "/home/vguttmann/.icestudio/venv/bin/apio"
Common.APP: Icestudio APP folder: /tmp/.mount_1fQc9L/usr/bin
common.APP_DIR: Icestudio execution folder: /tmp/.mount_1fQc9L/usr/bin

===> tools.CHECKTOOLCHAIN
>>>> utils.executeCommand => APIO_HOME_DIR="/home/vguttmann/.icestudio/apio" PATH="/home/vguttmann/.icestudio/venv/bin:/tmp/.mount_1fQc9L/usr/bin:/tmp/.mount_1fQc9L/usr/sbin:/home/vguttmann/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" "/home/vguttmann/.icestudio/venv/bin/apio" --version

>>(ERROR): /bin/sh: 1: /home/vguttmann/.icestudio/venv/bin/apio: not found

----!!!! ERROR !!!! -----
CMD: APIO_HOME_DIR="/home/vguttmann/.icestudio/apio" PATH="/home/vguttmann/.icestudio/venv/bin:/tmp/.mount_1fQc9L/usr/bin:/tmp/.mount_1fQc9L/usr/sbin:/home/vguttmann/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" "/home/vguttmann/.icestudio/venv/bin/apio",--version
  Error flag: true
  Error: true
------> MENU ENTRY POINT: Install Toolchain
[... 18 lines cut ...]
**** STEP: Install APIO
>>>> utils.executeCommand => "/home/vguttmann/.icestudio/venv/bin/pip3" install -U apio[blackiceprog,tinyfpgab,tinyprog,icefunprog]==0.8.4

>>(ERROR): /bin/sh: 1: /home/vguttmann/.icestudio/venv/bin/pip3: not found

----!!!! ERROR !!!! -----
CMD: "/home/vguttmann/.icestudio/venv/bin/pip3",install -U apio[blackiceprog,tinyfpgab,tinyprog,icefunprog]==0.8.4
**** STEP: APIO install oss-cad-suite
>>>> utils.executeCommand => APIO_HOME_DIR="/home/vguttmann/.icestudio/apio" PATH="/home/vguttmann/.icestudio/venv/bin:/tmp/.mount_1fQc9L/usr/bin:/tmp/.mount_1fQc9L/usr/sbin:/home/vguttmann/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" "/home/vguttmann/.icestudio/venv/bin/apio" install oss-cad-suite@0.0.8

>>(ERROR): /bin/sh: 1: /home/vguttmann/.icestudio/venv/bin/apio: not found

----!!!! ERROR !!!! -----
CMD: APIO_HOME_DIR="/home/vguttmann/.icestudio/apio" PATH="/home/vguttmann/.icestudio/venv/bin:/tmp/.mount_1fQc9L/usr/bin:/tmp/.mount_1fQc9L/usr/sbin:/home/vguttmann/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" "/home/vguttmann/.icestudio/venv/bin/apio",install,oss-cad-suite@0.0.8
**** FINAL STEP: Checking the installed apio
===> tools.CHECKTOOLCHAIN
>>>> utils.executeCommand => APIO_HOME_DIR="/home/vguttmann/.icestudio/apio" PATH="/home/vguttmann/.icestudio/venv/bin:/tmp/.mount_1fQc9L/usr/bin:/tmp/.mount_1fQc9L/usr/sbin:/home/vguttmann/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" "/home/vguttmann/.icestudio/venv/bin/apio" --version

>>(ERROR): /bin/sh: 1: /home/vguttmann/.icestudio/venv/bin/apio: not found

----!!!! ERROR !!!! -----
CMD: APIO_HOME_DIR="/home/vguttmann/.icestudio/apio" PATH="/home/vguttmann/.icestudio/venv/bin:/tmp/.mount_1fQc9L/usr/bin:/tmp/.mount_1fQc9L/usr/sbin:/home/vguttmann/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" "/home/vguttmann/.icestudio/venv/bin/apio",--version
  Error flag: true
  Error: true
  ```

cavearr, 2024-01-19:
Excuse me! i'm telling you when the error will be fixed.

Obijuan, 2024-01-19:
Please, make sure you have installed the required dependencies. It looks like you have missed one of them. Execute this command in the shell and try again
```
sudo apt install python3 python3-pip python3-venv xclip libfuse2
```

cavearr, 2024-01-19:
Try to download again! should be works

cavearr, 2024-02-05:
I'm closing this issue, i can't reproduce the error, probably an error for concrete user setup.

If problem persist after install the last wip (https://downloads.icestudio.io) reopen it.

Thanks!

closed 2024-02-05 by cavearr