FPGAwars/icestudio #716: Please include gtkwave in IceStudio distribution

zapta, 2024-02-14:
I am using IceStudio version below (couldn't not copy it as text for some reason).

[image]

The iverilog and vvp binaries are already included but gtkwave is not. Please include in the release also gtkwave, (unless if IceStudio already includes equivalent functionality).

cavearr, 2024-02-14:
Hi! Yes i'm preparing a new wip with it . Un the next days i'm telling you.

Your feedback on how you will use it should be very usefull for me. I'm working in supporting simulation capabilities and the way that you use it should be very important.

I'll updating the state of it in this thread.

zapta, 2024-02-14:
Thanks @cavearr, will look forward for it. My minimal need is to have gtkwave installed as part of icestudio so I don't need to struggle separately with its installation. Currently my simulation/flow of a block is as follows:

1. Verify and build the block in IceStudio GUI.
2. Exporting the testbench template and modifying it manually.
3. Running a shell script that runs iverilog, vpp, and gtkwave.

I am not sure how much work you plan doing on the testing/simulation part but I can help with you feedback.  Overall I think that it can be along these high level milestones:

1. Being able to run a single testbench and examining its waveforms all within icestudio's GUI.
2. A more ambitious goal of introducing the concept of a 'project', with library blocks and testbenches and being able to run all the project tests at once from the IceStudio GUI.

For goal #2, I think that most of the concepts can be borrows from software IDEs, where it's aware of a project (e.g. root folder), areas within the project (libraries, unit tests, configuration, etc) and can perform operations at a project level such as running all the unit tests. Since you are a software developer, you are familiar with those concepts and I think they should work just as well for HDL projects. Similar to unit tests, each testbench has artifacts such as log or waveform and user can examine them after the tests run.

Hope it helps.

cavearr, 2024-02-14:
Fully aligned!!! and thanks for your detailed explanation and ideas!.  In short i publish a new wip with the concept of projects (.iceprj like .ini of platformio) .

Thanks again, i'm telling you here soon.

zapta, 2024-02-14:
Thanks Carlos!

cavearr, 2025-01-25:
i'm closing this issue because i'm migrating new feature requests to Gihub discussions in **ideas** category and there are other  conversation here opened with this topic.

In few days this new feature will be released in Icestudio.

Thanks for all team!

closed 2025-01-25 by cavearr