FPGAwars/icestudio #718: Splash screen is shown even when disabled

zapta, 2024-02-14:
I checked the don't-display checkbox but the splash screen is still shown up when IceStudio starts.

[image]

cavearr, 2025-01-27:
@lmcapacho  the same, if you could take it ... ❤ 

I'm planning to modify this window to be more useful, if you fix this weird behaviour should be a solid start point for the next step.

lmcapacho, 2025-01-30:
Hi @cavearr !

I couldn't reproduce this bug. The splash screen only appears when switching between Icestudio versions, which is expected behavior. Could you provide more details about the issue?

zapta, 2025-01-30:
I installed a fresh icestudio a few weeks ago and it lands in the editor as expected.  (version 0.12 mac osx).

There is a temporary splash screen for a fraction of the second while icestudio loads but then it switch automatically to the editor.

cavearr, 2025-01-30:
hi @zapta! if only appears in osx, i'll put this thread in standby because i'm starting to generating new osx packages in short and we could the latest wips on this.

For the moment @lmcapacho  i think is better don't spent more time in this until we have clear the pattern to reproduce it.

zapta, 2025-01-30:
@cavearr, sounds good. I think it can also get closed. The splash appears just for a fraction of a second during loading and doesn't require the user to click to continue so it's not a problem any more.

jojo535275, 2025-01-31:
Hello, 
For the pattern, it seams this splash screen is wrongly displayed (at least ) every time we create a fully New .ice file. 
with  File/New :

https://github.com/user-attachments/assets/d0760f4d-3b60-4adc-9b1f-17a001d51556

cavearr, 2025-01-31:
Hi  @jojo535275 ! i'll take it! i could reproduce it randomly, i'll try to find the problem.

Thanks again!