FPGAwars/icestudio #719: Build fail with a trivial ICESugar-pro design

zapta, 2024-02-16:
I am trying to build for ICESugar-pro but all attempts fail during the build with a cryptic error message, even trivial ones. What am I doing wrong?  Details below.  Also, is there a way to get a proper error message from nextpnr-ecp5?

Icestudio version: 0.11.3w202402070902  (on Max OS 14.3, M2) 

Design:
[image]

Build error log
```
APIO_HOME_DIR="/Users/user/.icestudio/apio" PATH="/Users/user/.icestudio/venv/bin:/usr/bin:/bin:/usr/sbin:/sbin" "/Users/user/.icestudio/venv/bin/apio" build --board "iCESugar-Pro_(USB-Blaster)" --verbose-pnr -p "/Users/user/projects/fpga/repo/experimental/ice-build/test"

(DEBUG) Profile path: /Users/user/.icestudio/apio/profile.json
(DEBUG) Home_dir: /Users/user/.icestudio/apio

PATH: /Users/user/.icestudio/apio/packages/tools-oss-cad-suite/bin:/Users/user/.icestudio/apio/packages/tools-oss-cad-suite/lib:/Users/user/.icestudio/venv/bin:/usr/bin:/bin:/usr/sbin:/sbin

[Thu Feb 15 20:21:11 2024] Processing iCESugar-Pro_(USB-Blaster)
--------------------------------------------------------------------------------
yosys -p "synth_ecp5 -json hardware.json" -q main.v
nextpnr-ecp5 --25k --package CABGA256 --json hardware.json --textcfg hardware.config --lpf main.lpf --timing-allow-fail --force

Info: Logic utilisation before packing:
Info:     Total LUT4s:         0/24288     0%
Info:         logic LUTs:      0/24288     0%
Info:         carry LUTs:      0/24288     0%
Info:           RAM LUTs:      0/12144     0%
Info:          RAMW LUTs:      0/ 6072     0%

Info:      Total DFFs:         0/24288     0%

Info: Packing IOs..
Info: pin 'v46fc91$tr_io' constrained to Bel 'X72/Y17/PIOB'.
Info: pin 'v139d61$tr_io' constrained to Bel 'X72/Y14/PIOC'.
Info: Packing constants..
scons: *** [hardware.config] Error -11
========================= [ ERROR ] Took 1.42 seconds =========================
```

test.ice source file.
[test.ice.zip](https://github.com/FPGAwars/icestudio/files/14305934/test.ice.zip)

cavearr, 2024-02-19:
Hi @zapta ! sorry for the delay in the reponse.

I could build without problems:

[image]

Check for the latest wip at https://downloads.icestudio.io to assure have the latest toolchain installed:

[image]

You need to view 0.9.1 toolchain installed (Develop version of toolchain).

zapta, 2024-02-19:
Thanks Carlos.

cavearr, 2025-01-24:
This is fixed? @zapta

zapta, 2025-01-24:
@cavearr, does the zip below contains the correct apio project for this issue?  If not, can you sent it to me?

I tested ``apio build`` with the project in the zip file below and with the latest apio dev and it works.

[test.zip](https://github.com/user-attachments/files/18544032/test.zip)

```
~/Downloads/ice-build/test$ ls
apio.ini	main.lpf	main.v
~/Downloads/ice-build/test$ apio build
Setting the envinronment.
[Fri Jan 24 15:57:31 2025] Processing icesugar-pro-usb-blaster
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
yosys -p "synth_ecp5 -top main -json _build/hardware.json" -q main.v
nextpnr-ecp5 --25k --package CABGA256 --speed 6 --json _build/hardware.json --textcfg _build/hardware.config --report _build/hardware.pnr --lpf main.lpf -q --timing-allow-fail --force
ecppack --compress --db /Users/user/.apio/packages/oss-cad-suite/share/trellis/database _build/hardware.config _build/hardware.bit
================================================================================= [SUCCESS] Took 1.03 seconds =================================================================================
~/Downloads/ice-build/test$ 
```