FPGAwars/icestudio #721: Errores al verificar

ea3hmj, 2024-02-29:
Soy nuevo en el mundo de las fpga y una vez instalado y funcionando el softw icestudio he empezado a probar cosas sacadas de los cuadernos tecnicos.

Pero cuando cargo un ejemplo me da errores, voy bloque a bloque hasta que veo el error, pero me extrañana tanto que he decidido escribir a ver donde esta la solucion.

Como ejemplo de la coleccion Collection-Jedi-1.12.0, el bloque Serial-rx-block - v0.2.
[image]

SI entro en un bloque y verifico, me visualiza el error
[image]

Este error me pasa en todos los bloques y por lo que he visto el registro "q"! es el nombre mas comun.

Hay alguna opción de configuración para evirtar este error o ¿como puedo solucionarlo? porque en ejemplos largos da en un monton de bloques.

Saludos
Jose

cavearr, 2024-03-01:
Hola! te cuento, esos bloques están desactualizados, con las últimas versiones de la toolchain cambia ligeramente el "como implementar el hardware" de algunos módulos.

Mira a ver si hubiera alguna actualización en los repositorios de Obijuan y sino la solución es relativamente sencilla. Hoy por la tarde intento migrarte ese ejemplo para que lo tengas como "plantilla" el cambio como te digo es muy sencillo pero hay que implementarlo en todos los bloques.

cavearr, 2024-03-18:
Cierro la issue al no tener noticias y ser un problema conocido y que ya está en solución por parte de los desarrolladores de las colecciones. 

Cierro la issue, si vuelves a necesitar algo , reábrela.

closed 2024-03-18 by cavearr