FPGAwars/icestudio #723: Icestudio error message "Board iceWerx not connected"

ssjimh, 2024-03-06:
Icestudio V0.11,  Board selected is "iceWerx"

When I connect the iceWerx iCE40-HX8K board to USB, Windows 10 Device Manager shows "libusb-win32 devices, iceFUN" 
Board power LED is on, other led is blinking as per new unaltered board.

I select Icestudio Menu: Tools, Drivers, Enable, then read "Serial driver installation instructions" and click OK

Window appears:
Virtual Serial Driver 
Windows 10 USB Serial Device driver is present
No installation is necessary

I unplug and reconnect the Icewerks board.  Windows makes the usual USB connection sounds.

I select Menu: Tools, Upload

Error Message:  "Board iceWerx not connected"

I have removed the windows driver "libusb-win32 devices, iceFUN" and started over, same results.

What should I try next?

ssjimh, 2024-03-07:
Same behavior with the iceFUN iCE40HX8K board.

cavearr, 2024-03-07:
Hi! I'm going to investigate why it broke, I'll keep you posted, I'll probably ask you to do some tests.

ssjimh, 2024-03-07:
Thank you.  

I see in Win 10 Device Manager libusb-win32 devices / iceFUN driver, appears when I plug in the iceFUN or iceWerx board.

ssjimh, 2024-03-08:
Got it!

I installed icestudio on my laptop and it went well, bulding and downloading to iceFUN board properly.
Used this version:   [icestudio-0.11.3w202402041002-Windows64bits x86.zip]

On my Desk PC, I scrubbed previous install completely and re-installed as on laptop.  Driver still wouldn't install, "board not connected".   No com port assignment showed in Windows 10 device manager.

I completely disabled and removed the "libusb-win32 devices, iceFUN" driver, uninstalled and scrubbed the icestudio file structure in c:\program files, and reinstalled.

During the  Tools, Drivers, Enable  process I still got this window:
"Windows 10 USB Serial Device driver is present, No installation is necessary"
But the com port was assigned and now can download to the iceFUN board!

Thanks, Jim

cavearr, 2024-03-08:
Thanks for the feedback!! I was trying to replicate your problem but couldn't.

I'm so happy you made it! I'm sure other users will find your experience very helpful.

I close the issue and if you have any other problems related to this, reopen it.

Thank you for using Icestudio!

closed 2024-03-08 by cavearr

ssjimh, 2024-03-09:
Hi Carlos,

What forum should I ask general questions about learning the best use of icestudio?

Thanks, Jim

Closed #723 [https://github.com/FPGAwars/icestudio/issues/723] as completed.

cavearr, 2024-03-09:
We are trying to create a great community in https://groups.google.com/g/fpga-wars-explorando-el-lado-libre appears an spanish group because was its origins but now we try to write in english.

Tell us all of you want!