FPGAwars/icestudio #724: IceStudio stuck on windows 10

amin-mm-dev, 2024-03-08:
I got the latest Icestudio release for Windows, but I can't get it to run after installing. It just keeps loading indefinitely, even after trying different versions and turning off AVG and Windows Defender.

I also granted administrator rights.

[image]

cavearr, 2024-03-08:
What version are you installing? Try the first in the table for your operating system at https://downloads.icestudio.io/

amin-mm-dev, 2024-03-08:
[image]

amin-mm-dev, 2024-03-08:
I installed that version in the previous screenshot

closed 2024-11-11 by cavearr