FPGAwars/icestudio #725: How are verilog modules implemented in icestudio?

ssjimh, 2024-03-10:
I found some sample verilog that I want to implement on the iceFUN board.   

I create a code block and paste in the verilog, which has module and endmodule heater and footers, which generate error messages.

I have extracted the body of two modules and created two code blocks and wired them together, and it works.  But I suspect there is a proper way to use modules.  Can two or more modules exist in one code block?

ssjimh, 2024-03-10:
Are Varilog variable names local to modules?  In a code block with multiple modules?

ssjimh, 2024-03-10:
Is there a log file or output file that the "builder" or "uploader" creates that has details about build errors?
Some error message pop-ups aren't very detailed, so I'm hoping to find details about the error.

Also, sometimes when I press "upload", it works for a few seconds and finishes without error messages, but the design is not uploaded to the board.  How can I find out why?

Thanks!

cavearr, 2024-03-10:
Hi @ssjimh ! each icestudio block is a verilog module, but you could try to define a module inside the block (the scope should works only into this code block).

For the moment you could import verilog files that are in your project folder. For the moment folders are not supported (but very soon support it, i'm working on it!).
For this, you only put a comment at the start of the verilog editor:

// @include ram_sdp.v

ram_sdp.v for example should be in the same folder that your .ice

[image]

About the yosys output, you could find in this option:

[image]

Greetings!

ssjimh, 2024-03-14:
Thanks Cavearr!

closed 2024-03-14 by ssjimh