FPGAwars/icestudio #730: md.cpython-39-darwin.so cannot be opened on MacOS fresh install

fxframes, 2024-04-14:
Not much to add, Icestudio installs fine on macOS Sonoma 14.4.1 but as soon as it's opened it throws this error.
I found the file and tried to do the MacOS right-click thing and open it "manually" using Icestudio but to no avail.
Thanks for any insights.

[image]

cavearr, 2024-04-14:
What version of Icestudio are you using?

fxframes, 2024-04-14:
Thanks for the quick reply.
The file I downloaded from icestudio.io is named icestudio-0.11.0-rc2-osx64.dmg.
I'm on an M1 Mac by the way.

cavearr, 2024-04-14:
try downloading the last wip from https://downloads.icestudio.io

Our wips are very stable and more up to date that our stable versions that are used by some universities that need very stability.

Tell me if this works!

fxframes, 2024-04-15:
Thanks, I tried the two latest MacOS ARM versions available but both give me an error.

[image]

cavearr, 2024-04-15:
Give me a couple of days to fix it, there are some problem with osx signature

fxframes, 2024-04-16:
Thank you for your support. =)

setrar, 2024-04-20:
I uninstalled the homebrew version and I was able to install this version [icestudio-0.11.3w202402070902-OSXARM 64.dmg](https://api.github.com/repos/FPGAwars/icestudio/actions/artifacts/1364134182/zip)

[image]

cavearr, 2024-06-06:
Hello everyone! At the moment the issue seems to be fixed, now packages are automatically signed in github CI actions and the user only needs to accept the unknown developer in the security check in the security section of the settings panel (as it usually appears with other apps).

I'm working on a dmg with less friction, but at the moment it works relatively comfortably.

I'll be closing the thread and reopening it in the near future to let you know about the new improvements to damage output.

Greetings and thank you very much for using Icestudio!

grijspetje, 2024-06-06:
I just checked and the last update is from 5th of june. Is version 0.11.3w202406050906 the last version because  this version has the python install problem.

kind regards,

Peter

cavearr, 2024-06-06:
Hi @grijspetje !, how do you install python in your system? i can't reproduce. the python problem and i think could be a python environment problem that i want to fix or patch.

For the screenshot i view the problem appears when install the toolchain ,isn't it?

But in my OSX when try to install the toolchain appears the next dialog, asking for admin user and password:

[image]

In your case don't appear this? if not appear i'll need to check why, could be possible that i have installed some package that is not standar (development package).

Thanks for your feedback!

grijspetje, 2024-06-06:
Carlos,

I send you some screenshots to show yoou what I encounder.

Downloaded icestudio-0.11.3w202406050906-osxarm64 and started up:

￼

download the latest tools. 

No problem

Installing drivers:

takes a long time

￼

and now it seems to be ok!

I dont know what has changed.

kind

Regards,

Peter

cavearr, 2024-06-06:
Hello Peter! I don't know what could happen, I was never able to reproduce it, it could be some version overlap in your installation, I really don't know.

If it already works and looks good to you, I'll close the issue and if it appears again, reopen it, okay?

Thanks again for your patience and your comments!

closed 2024-06-06 by cavearr

grijspetje, 2024-06-06:
OkMet vriendelijk groeten,Peter van den BosVerstuurd vanaf mijn iPad