FPGAwars/icestudio #732: nandland GO board and windows 11

DominionSoftware, 2024-04-18:
The Enable drivers feature fails silently. My device is seen by windows 11 as USB Serial Converter A and USB Serial Converter B. 
I select Enable and the Dialog box comes up saying how to install the drivers, but when I click ok nothing happens.

I tried running as Administrator as well. This is with Nightly build and Stable build.

cavearr, 2024-04-19:
Hi! we are changing a lot of things in the next weeks, and now are some things "unstable".

Download the latest icestudio wip from https://downloads.icestudio.io , install the development apio version (you need install git).

When enable drivers in zadig select WinUSB instead libusb.

Try it and tell me!

greenliquidlight, 2024-05-05:
I just did a fresh install using the latest icestudio and can confirm that my Nandland Go board is working.  I converted a simple LED + PushButton example and loaded it to the board without any issues.
Using the WinUSB driver for interface0 worked for me.

cavearr, 2025-01-19:
Thanks a lot for your feedback, sorry for my delay in the response!

closed 2025-01-19 by cavearr