FPGAwars/icestudio #740: v0.11 Linux installation missing menu options

bgalport, 2024-05-14:
Just installed the latest version (0.11) on Ubuntu 24.04 and find that the only the Basic menu option is displayed. The Bit, Logic and other options I see on web examples are missing. I've tried loading other collections but nothing changes. I am able to load all the example files. Have I missed something?

[image]

bgalport, 2024-05-14:
Hi @cavearr, I used the Tools->Collections->Add option to add Collection-Jedi. Even the default collection should have included logic blocks shouldn't it?

bgalport, 2024-05-14:
Thank you! I would never have thought of looking there. I can now see all the available blocks. 

[image]

cavearr, 2024-05-14:
You are welcome!

closed 2024-05-14 by cavearr