FPGAwars/icestudio #748: IceStudio 0.11/Ubuntu 24.04: TinyFPGA-BX board not connected

bgalport, 2024-05-27:
My TinyFPGA-BX board appears in _lsusb_ as the following.
`Bus 001 Device 008: ID 1d50:6130 OpenMoko, Inc.`

I can build and upload code using command-line apio after having enabled the driver with the command:
`apio drivers --serial-enable`

I also had to add my user to the 'dialout' group so I could write to the port.

In icestudio ran the 'Enable Driver' option and can see a _80-fpga-fdti.rules_ file that ooks like this:
```
ATTRS{idVendor}=="0483", ATTRS{idProduct}=="5740", ENV{ID_MM_DEVICE_IGNORE}="1"
# Disable ModemManager for TinyFPGA B2
ATTRS{idVendor}=="1209", ATTRS{idProduct}=="2100", ENV{ID_MM_DEVICE_IGNORE}="1"# Disable ModemManager for TinyFPGA BX
ATTRS{idVendor}=="1d50", ATTRS{idProduct}=="6130", ENV{ID_MM_DEVICE_IGNORE}="1"# Disable ModemManager for iceFUN
ATTRS{idVendor}=="04d8", ATTRS{idProduct}=="ffee", ENV{ID_MM_DEVICE_IGNORE}="1"
```

But when I try to upload from icestudio I get a 'board TinyFPGA-BX not connected' error

```
APIO_HOME_DIR="/home/bruce/.icestudio/apio" PATH="/home/bruce/.icestudio/venv/bin:/tmp/.mount_bLTI5u/usr/bin:/tmp/.mount_bLTI5u/usr/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin" "/home/bruce/.icestudio/venv/bin/apio" upload --board TinyFPGA-BX --verbose-pnr -p "/tmp/icestudio-96127nNzXiEuiSYz"

(DEBUG) Profile path: /home/bruce/.icestudio/apio/profile.json
(DEBUG) Home_dir: /home/bruce/.icestudio/apio
(DEBUG) Profile path: /home/bruce/.icestudio/apio/profile.json
(DEBUG) Home_dir: /home/bruce/.icestudio/apio
(DEBUG) Run Command: lsusb
(DEBUG) System_base_dir: /home/bruce/.icestudio/apio/packages/tools-oss-cad-suite
(DEBUG) System bin dir: /home/bruce/.icestudio/apio/packages/tools-oss-cad-suite/bin
(DEBUG) Executable file: /home/bruce/.icestudio/apio/packages/tools-oss-cad-suite/bin/lsusb
Error: board TinyFPGA-BX not connected 
```

The [documentation](url) says you can also enable the driver from apio using the command:
`apio drivers --ftdi-enable`
 
Which is different to the option I used with command-line apio where I specified _serial_ rather than _ftdi_.

Any help it getting this setup would be appreciated.

cavearr, 2024-06-04:
Hi! @bgalport in this last post of this fpgawars thread:

https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/hbWVxvIWU_E/m/V2itMDZpAAAJ

i'm sending a replacement for tinyfpga tool ¿could you try it ant tell me if it works for you?

bgalport, 2024-06-04:
Hi @cavearr 
That works for me! I also have one of the crowdsource boards with missing meta data and I can see from your code how you use hard coded meta data if none is found on the device. Will this fix be applied to the command line apio disribution as well?
Thanks for your efforts

cavearr, 2024-06-04:
thanks a lot for your feedback!! yes i apply to Apio distribution today or tomorrow at last, i'm waiting for some users responses to give for valid the fix.

I'm maintain open the issue until apio will be fixed.