FPGAwars/icestudio #751: version 0.12 - additional support for macOS intel?

suschizu, 2024-06-22:
Will there also be a version for Intel Macs?
the download 
https://github.com/FPGAwars/icestudio/releases/download/v0.12/icestudio-0.12-osxarm64.dmg  
contains an ARM-only version.
thanks!

cavearr, 2024-06-23:
Hi! first of all, we don't had beed feeback from users  with intel osx and we can't do tests for it for the new versions.

About the Rosetta. framework, it has a very bad performance for nwjs engine.

If you could give me  feedback for osx intel  tests i'll working in the dmg and use you as testers.

Are you ready for the tests?? ;)  Give me a couple of days and prepare the dmg bundle for intel osx, i'm telling you in this thread ok?

placessweden, 2024-06-23:
I'm very happy if I can help you testing! Today I have messed around with the source with grunt-cli and npm (which I've never used in my life :) which give me two dependency errors, grunt-wget and grunt-appdmg

cavearr, 2024-06-23:
If you need help tell me.

To start icestudio from cli from the sources if you need help, tell me.

You only need to execute:

npm install --legacy-peer-deps

and when finish it:

npm start

placessweden, 2024-06-23:
Thankyou! Works like a charm on my Intel-Mac. However, there are complaints that grunt-appdmg is missing.
[image]

cavearr, 2024-06-23:
Yes, don't worry, this is one of the problems to generate the intel dmg. I'm working on updating all the packages because we use some "abandoned" packages, and I'm working on a lot of breaking changes and these packages are going away soon.

Thank you and in the next few days I will tell you to try the intel dmg package. I need your comments, so I can stabilize it. Many other users will be very grateful to you!