FPGAwars/icestudio #753: Unable to upload to TinyFPGA BX on macOS

Bexin3, 2024-06-24:
Hello, I tried to upload a totally basic schematic with one input one output and some others, but I keep getting an error. With all other boards it just says the board isn't connected, but with BX I get the following error; 
I tried using both the 0.9.5 and 0.9.6 toolchains but the result is the same.
Info: No apio.ini file
Info: No apio.ini file
[Mon Jun 24 10:14:47 2024] Processing TinyFPGA-BX
--------------------------------------------------------------------------------
tinyprog --libusb --program hardware.bin
Traceback (most recent call last):
File "/Users/redacted/.icestudio/venv/bin/tinyprog", line 8, in 
sys.exit(main())
File "/Users/redacted/.icestudio/venv/lib/python3.9/site-packages/tinyprog/__main__.py", line 249, in main
active_boards = get_ports(device) + get_ports("1209:2100")
File "/Users/redacted/.icestudio/venv/lib/python3.9/site-packages/tinyprog/__init__.py", line 39, in get_ports
for d in usb.core.find(idVendor=vid, idProduct=pid, find_all=True)
File "/Users/redacted/.icestudio/venv/lib/python3.9/site-packages/usb/core.py", line 1309, in find
raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
scons: *** [upload] Error 1

Bexin3, 2024-06-24:
I tried first file in https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/hbWVxvIWU_E/m/V2itMDZpAAAJ?pli=1 but no luck

Bexin3, 2024-06-24:
For further information this happens no matter if the board is connected and I press the button or even if its fully disconnected

cavearr, 2024-06-24:
hi! when do you buy your board? the month and year is important to check what firmware is in it.

your machine is intel or arm based?

Bexin3, 2024-06-24:
> hi! when do you buy your board? the month and year is important to check what firmware is in it.
> 
> 
> 
> your machine is intel or arm based?

Hello, the machine is arm based. In terms of when it was in October of 2023 so one of the new batches.

cavearr, 2024-06-24:
Thanks , could you help me to try to fix it? my tiny is prior to this and works fine, if you could give me feedback , we could find a solution.

You tried the last __init_.py in the thread? (you mention it, but i don't understand if you overwrite with the last __init__.py) https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/hbWVxvIWU_E/m/V2itMDZpAAAJ?utm_medium=email&utm_source=footer

Bexin3, 2024-06-24:
> Thanks , could you help me to try to fix it? my tiny is prior to this and works fine, if you could give me feedback , we could find a solution.
>
>
>
> You tried the last __init_.py in the thread? (you [...]

Yes I have tried it but unfortunately it didn't work and had no effect overall.

cavearr, 2024-06-24:
Ok! give me some time to prepare the new fix and tell you to try it.

cavearr, 2024-07-02:
Hi @Bexin3  ! 

Ill put you three files here [https://mega.nz/folder/jM9DlZhQ#7f9N9uniTnS3MHvS5mF3HQ](https://mega.nz/folder/jM9DlZhQ#7f9N9uniTnS3MHvS5mF3HQ)

- page1.txt and page2.txt are the missing metada in your fpgatiny.
- tinyprog a patched version of tinyprog for osx arm
-test-tiny.bin a blinking light test if you dont have a test bitstream.

remember  first do in your shell:

 chmod 755 tinyprog 

 i need you do this steps in a shell to know if it works:

tinyprog --atb --security page1.txt -a 2
tinyprog --atb --security page2.txt -a 3

And then try to upload some bitstream:

tinyprog --atb -p hardware.bin

hardware.bin is your test bitstream

Tell me if it works!

Bexin3, 2024-07-03:
> Hi @Bexin3 !
>
> Ill put you three files here https://mega.nz/folder/jM9DlZhQ#7f9N9uniTnS3MHvS5mF3HQ
>
> * page1.txt and page2.txt are the missing metada in your fpgatiny.
> * tinyprog a patched version of [...]

Hello, so far no luck. Not sure if I am doing this correctly but I downloaded the folder, changed to the folder, ran the first command. Then trying to run the second it said zsh: command not found: tinyprog, so I instead tried to run it as ./tinyprog, in which case it seems to just run the executable and it doesn't change, and neither after running these order does the outcome, saying 
    TinyProg CLI
    ------------
    Using device id 1d50:6130
    No port was specified and no active bootloaders found.
    Activate bootloader by pressing the reset button.
Which to clarify I did press the button

cavearr, 2024-07-03:
Sorry maybe i write you very fast XD

Do this steps in order:

1) Downdload de folder content
2) Enter in it
3) chmod 755 tinyprog
4) Connect your tinyfpga
4) ./tinyprog --atb --security page1.txt -a 2 
5) ./tinyprog --atb --security page2.txt -a 3
6) ./tinyprog --atb -p hardware.bin

If you send me the output of the 4) 5) and 6) steps , i'm very gratefully.

Bexin3, 2024-07-03:
Outputs are as follows;
5)
    TinyProg CLI
    ------------
    Using device id 1d50:6130
    No port was specified and no active bootloaders found.
    Activate bootloader by pressing the reset button.
6)

    TinyProg CLI
    ------------
    Using device id 1d50:6130
    No port was specified and no active bootloaders found.
    Activate bootloader by pressing the reset button.
7)
    TinyProg CLI
    ------------
    Using device id 1d50:6130
    No port was specified and no active bootloaders found.
    Activate bootloader by pressing the reset button.

cavearr, 2024-07-03:
Have you brew installed? if not install it ( https://brew.sh/ ) and then execute the next two commands:

brew install libusb
ln -s /opt/homebrew/lib ~/lib

When this was done, execute again your first test :

./tinyprog --libusb --program hardware.bin

try first with the original tinyprog and if not works try with the patched.

Bexin3, 2024-07-03:
> Have you brew installed? if not install it ( https://brew.sh/ ) and then execute the next two commands:
>
> brew install libusb ln -s /opt/homebrew/lib ~/lib
>
> When this was done, execute again your [...]

I did have both installed, I reinstalled it but no luck, both original and new still have the same output.

cavearr, 2024-07-03:
had you did the command ?:

`
ln -s /opt/homebrew/lib ~/lib

`

Bexin3, 2024-07-03:
> had you did the command ?:
> 
> ` ln -s /opt/homebrew/lib ~/lib
> 
> `

Yeah, no difference

cavearr, 2024-07-03:
Coud be obvious but you have pyusb installed?

pip install pyusb

cavearr, 2024-07-03:
And if the previous not work try the next command before executing tinyprog:

export DYLD_LIBRARY_PATH="/opt/homebrew/lib:$DYLD_LIBRARY_PATH"

Bexin3, 2024-07-03:
> Coud be obvious but you have pyusb installed?
> 
> pip install pyusb

Not sure if I had it tbh, to be clear I installed the toolchain through the app and didn't do any terminal installations before now so may lack something else if there are other requirements.
I tried it and with the export thing but the behaviour didnt change

cavearr, 2024-07-03:
with the tiny connected , could you do lsusb and send me the output?

Bexin3, 2024-07-03:
> with the tiny connected , could you do lsusb and send me the output?

Bus 000 Device 001: ID 1d6b:XHCI
XHCI
XHCI Linux Foundation USB 3.1 Bus

Bexin3, 2024-07-03:
I get the same thing when the board is not connected though weird

Bexin3, 2024-07-03:
Looking at TinyFPGA page looks like multiple people are reporting issues with the new batch with no response from the developer..

cavearr, 2024-07-04:
Yes for this, i'm asking you about the date of your buy.

I think you have some problems. The first one is your system not detect the board (this is until tinyprog works).

lsusb should detect the board first. I thinkt some test, tell you soon.

Bexin3, 2024-07-04:
> Yes for this, i'm asking you about the date of your buy.
> 
> 
> 
> I think you have some problems. The first one is your system not detect the board (this is until tinyprog works).
> 
> 
> 
> lsusb should detect the board first. I thinkt some test, tell you soon.

Good news, after cutting of a pin of the voltage regulator the board is now detected, will have a look if your programs work now

Bexin3, 2024-07-04:
Sadly your program still doesn't upload, but lsusb now gives the following output;
Bus 002 Device 001: ID 1d50:6130 1d50 Communication Device 
Bus 000 Device 000: ID 1d50:6130 1d50 USB 3.1 Bus

cavearr, 2024-07-04:
The cut of the pin is some solution that you view in some place?

The original and patched  have the same output?

Bexin3, 2024-07-04:
> The cut of the pin is some solution that you view in some place?
> 
> 
> 
> The original and patched  have the same output?

Yeah the cut pin is from a thread where it turns out the voltage regulator on the new batch was wrongly placed and produced 5V instead of 3.3v which is why the computer woudlnt detect the board and I had no output. Now I have the output from the lsusb as what I sent detecting the board though I still can't upload to it; I can't test the original output as I don't have one but the ID seems to be correct

cavearr, 2024-07-04:
i'm refering if you try the original tinyprog and patched

Bexin3, 2024-07-04:
> i'm refering if you try the original tinyprog and patched

Ahh, both of them have the same output sadly

cavearr, 2024-07-04:
Could you try the original tinyprog but overwritting the files in my last post in this thread? https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/hbWVxvIWU_E/m/V2itMDZpAAAJ?pli=1

Bexin3, 2024-07-06:
> Could you try the original tinyprog but overwritting the files in my last post in this thread? https://groups.google.com/g/fpga-wars-explorando-el-lado-libre/c/hbWVxvIWU_E/m/V2itMDZpAAAJ?pli=1

tried it, still getting the no api.ini file error

Bexin3, 2024-07-06:
Maybe this could help https://github.com/tinyfpga/TinyFPGA-Bootloader/pull/73

Bexin3, 2024-07-21:
Further to this trying to use icestudio on windows I just get tinyfpga not connected error even though it is

Bexin3, 2024-07-21:
This is the output of default tinyprog

```
PS C:\Users\Bexin\Downloads\tinyfpga\tinyfpga> tinyprog -p test-tiny.bin

    TinyProg CLI
    ------------
    Using device id 1d50:6130
    Only one board with active bootloader, using it.
    Programming COM4 with test-tiny.bin
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Bexin\AppData\Local\Programs\Python\Python312\Scripts\tinyprog.exe\__main__.py", line 7, in <module>
  File "C:\Users\Bexin\AppData\Local\Programs\Python\Python312\Lib\site-packages\tinyprog\__main__.py", line 334, in main
    addr = fpga.meta.userimage_addr_range()[0]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Bexin\AppData\Local\Programs\Python\Python312\Lib\site-packages\tinyprog\__init__.py", line 182, in userimage_addr_range
    return self._get_addr_range(u"userimage")
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Bexin\AppData\Local\Programs\Python\Python312\Lib\site-packages\tinyprog\__init__.py", line 188, in _get_addr_range
    addr_str = self.root[u"bootmeta"][u"addrmap"][name]
               ~~~~~~~~~^^^^^^^^^^^^^
TypeError: 'NoneType' object is not subscriptable
```

cavearr, 2024-07-21:
Hi @Bexin3!, have you tried to overwrite the python files in your windows installation? i think the not connection problem disappear but you can't program the board. For the moment there are no solution for this batch of boards, that i know.

Bexin3, 2024-07-21:
> Hi @Bexin3!, have you tried to overwrite the python files in your windows installation? i think the not connection problem disappear but you can't program the board. For the moment there are no solution for this batch of boards, that i know.

Hello, not yet but I can certainly try, how would this be done on windows?

Bexin3, 2024-07-21:
> Hi @Bexin3!, have you tried to overwrite the python files in your windows installation? i think the not connection problem disappear but you can't program the board. For the moment there are no solution for this batch of boards, that i know.

Hello, I now tried to just overwrite the init file as per original instructions and it worked for windows!

cavearr, 2024-07-21:
Yeah! :) and you could program de tiny?

Bexin3, 2024-07-21:
> Yeah! :) and you could program de tiny?
> 
> 
> 
> Thanks a lot for your feedback! now i can put this fix for all (until your  tests  i can't know if it works!)
> 
> 
> 
> Please tell me if all works (at OSX you tell me that this fix not works , correct?)

You're welcome, thanks for making the fix!! Yes, I tried the test program and then modified it a bit and both time I was able to upload and the led blinked as expected!! 
On macOS sadly this doesn't seem to work, but I'll give it another try later just in case I did something wrong, though there the error isn't just with uploading so unlikely to work

cavearr, 2024-07-21:
We are a team! if you modify with some improvement please paste here , your changes will be welcome!

Patronics, 2024-08-20:
I seem to have the same or similar issue on MacOS, here's the output it gives me:

```
APIO_HOME_DIR="/Users/username/.icestudio/apio" PATH="/Users/username/.icestudio/venv/bin:/usr/bin:/bin:/usr/sbin:/sbin" "/Users/username/.icestudio/venv/bin/apio" upload --board TinyFPGA-BX --top-module main -p "/Applications/icestudio.app/Contents/Resources/app.nw/resources/collection/examples/1. Basic/ice-build/01. One LED"

Info: No apio.ini file
[Tue Aug 20 13:10:46 2024] Processing TinyFPGA-BX
--------------------------------------------------------------------------------
tinyprog --libusb --program hardware.bin
Traceback (most recent call last):
File "/Users/username/.icestudio/venv/bin/tinyprog", line 8, in <module>
sys.exit(main())
File "/Users/username/.icestudio/venv/lib/python3.9/site-packages/tinyprog/__main__.py", line 249, in main
active_boards = get_ports(device) + get_ports("1209:2100")
File "/Users/username/.icestudio/venv/lib/python3.9/site-packages/tinyprog/__init__.py", line 40, in get_ports
for d in usb.core.find(idVendor=vid, idProduct=pid, find_all=True)
File "/Users/username/.icestudio/venv/lib/python3.9/site-packages/usb/core.py", line 1309, in find
raise NoBackendError('No backend available')
usb.core.NoBackendError: No backend available
scons: *** [upload] Error 1
========================= [ ERROR ] Took 0.22 seconds =========================

```
Replacing that __init__ file as described by OP didn't seem to have any effect either. Similarly, running your tinyprog file with the arguments you provided (`--atb -p test-tiny.bin`) produced the same output as described in the original issue. 

*However*, if I added the --pyserial argument [as suggested in the update here](https://electronics.stackexchange.com/questions/662026/tinyprog-has-issues-programming-the-tinyfpga-bx-module-on-macos-m1), it gets a bit farther, giving the output 

```
./tinyprog --pyserial --atb -p test-tiny.bin

    TinyProg CLI
    ------------
    Using device id 1d50:6130
    Only one board with active bootloader, using it.

    Programming /dev/cu.usbmodem1101 with test-tiny.bin
usage: tinyprog [-h] [--version] [-l] [-p PROGRAM] [-u PROGRAM_USERDATA]
                [--program-image PROGRAM_IMAGE] [-b] [-c COM] [-i ID]
                [-d DEVICE] [-a ADDR] [-m] [--security SECURITY]
                [--update-bootloader] [--libusb] [--pyserial] [--atb]
tinyprog: error: unrecognized arguments: -B -S -I
Traceback (most recent call last):
  File "tinyprog/__main__.py", line 524, in <module>
  File "tinyprog/__main__.py", line 410, in main
  File "tinyprog/__init__.py", line 272, in userimage_addr_range
  File "tinyprog/__init__.py", line 279, in _get_addr_range
AttributeError: 'NoneType' object has no attribute 'get'
[92073] Failed to execute script '__main__' due to unhandled exception!
multiprocessing/resource_tracker.py:96: UserWarning: resource_tracker: process died unexpectedly, relaunching.  Some resources might leak.
usage: tinyprog [-h] [--version] [-l] [-p PROGRAM] [-u PROGRAM_USERDATA]
                [--program-image PROGRAM_IMAGE] [-b] [-c COM] [-i ID]
                [-d DEVICE] [-a ADDR] [-m] [--security SECURITY]
                [--update-bootloader] [--libusb] [--pyserial] [--atb]
tinyprog: error: unrecognized arguments: -B -S -I
```
(I wasn't sure what your two security page files were or if those were necessary, so skipped that step for now, if that's a factor at all)

Patronics, 2024-08-20:
I was able to successfully program the tinyfpga using your test-tiny.bin file along with a separately installed copy of tinyprog though, which was produced with the commands:

```
python -m venv venv
source ./venv/bin/activate
pip install apio tinyprog
apio install system scons iverilog
apio drivers --serial-enable 
```
This used python 3.12.5, while I noticed that the python version that icestudio used was python 3.9, if that's important? 
Here's the output of PIP freeze, in case the versions that are different from your version and are important: 

```
apio==0.9.5
attrs==24.2.0
certifi==2024.7.4
charset-normalizer==3.3.2
click==8.1.3
colorama==0.4.6
idna==3.7
intelhex==2.3.0
jsonmerge==1.9.2
jsonschema==4.23.0
jsonschema-specifications==2023.12.1
packaging==24.1
pyserial==3.5
pyusb==1.2.1
referencing==0.35.1
requests==2.28.2
rpds-py==0.20.0
SCons==4.2.0
semantic-version==2.9.0
setuptools==73.0.1
six==1.16.0
tinyprog==1.0.21
tqdm==4.66.5
urllib3==1.26.19
wheel==0.44.0
```

cavearr, 2024-08-20:
Thanks @Patronics !! 

Thanks for the feedback and for trying everything. Since my tinyfpga works fine, all my attempts to help are blind, your help is essential.

The python version can be 3.12 without a problem, in fact i should update it shortly at least in the WIPS.

I am going to calmly reread what you have tried and what has worked for you to see if we can close a final version to include it in icestudio so that it works for all users. Can I count on you for the tests?

Thank you so much!

Patronics, 2024-08-20:
Yeah, sure, I'm happy to help with testing, @cavearr! Let me know whenever you have something ready to test, or any other questions :)