FPGAwars/icestudio #767: Ctrl+U opens index.html in a text editor

colltoaction, 2024-08-31:
Hi! I'm seeing an `index.html` editor window every time I hit Ctrl+U. This doesn't happen with the Upload menu button or other shortcuts such as Ctrl+B.

I think the issue comes from unfinished work around here, or maybe with the shortcut settings.

https://github.com/FPGAwars/icestudio/blob/63c6c71a3c705f88ff289935f129ffdb34072ea3/app/scripts/services/utils.js#L1100-L1109

I'm on Icestudio 0.12 on Linux.

Thanks for taking a look, maybe I can fix it with some guidance.

cavearr, 2024-08-31:
Hi Martin! could you attach an screenshot with the editor window opened after Ctrl+U?

I can't reproduce your problem and i need to view how is happening to fix it or help you to fix it.

Thanks!

colltoaction, 2024-08-31:
Sure!

[image]

cavearr, 2024-08-31:
WTF!!! 😮 i think you have some shortcut in your system linked to open an external editor. I'll try to research it and tell you.

what's your linux distribution? 

could you try if in Chrome browser and in firefox if you press ctrl+u in any page the editor appears?

Thanks!

colltoaction, 2024-08-31:
😂 I'm using the AppImage in Ubuntu 23.04 with Firefox and I have not installed Chrome. The firefox shortcut for view source is Ctrl+U as well.

cavearr, 2024-08-31:
Yes i think the Operating system takes de control over ctrl+U for open editor in each application that could be editable XD

I look for a solution to this, thanks for detect it! it's very weird, i'm telling you soon

colltoaction, 2024-08-31:
Thanks! Tengo unas cositas más de UX si querés que te pase pero mirá que soy muy detallista jaja.

cavearr, 2024-08-31:
Todo el feedback es bueno tu lanza ideas se evalúan y se mira a ver. Yo ando preparando un gran set de cambios a nivel de ui y motor de cara al otoño, puede ser buen momento para nuevas ideas por si pudieran encajar.

TimRudy, 2024-09-01:
I noticed that earlier too - on Windows/Chrome - I didn’t research it, but yes, it would be interesting to fix it and understand what the cause was.

cavearr, 2024-09-01:
It appears that this shortcut is used by the operating system to open the code editor configured on the system by default (if you have installed and configured-linked the editor on the system). I haven't found any solution yet, I think it might be impossible to replace this (I have to try some ideas tomorrow but I think it might be a limitation). 

If the limitation is confirmed, I think it might be a good idea to change this shortcut (e.g. Ctrl+Shit+U or WindowsKey+U).

closed 2025-01-08 by cavearr