FPGAwars/icestudio #768: Comentarios generales sobre UX

colltoaction, 2024-08-31:
Hola! Quería dejar algunos comentarios de UX que tengo como nuevo usuario. Todo anda muy bien por suerte!

### touch

Me sorprendió cuando se me resbaló el dedo sobre el monitor y las cajas se movieron muy bien. Se puede dar soporte a otras interacciones como zoom.

### menúes
 
Las notificaciones desaparecen muy rápido y no se pueden revisar.

Collections manager y Basic en el menú principal están desconectados conceptualmente. CM es muy limitado, por ejemplo no tiene "expandir todo".

El menú izquierdo podría incluirse en el principal para reducir la cantidad de reglas de interacción.

### otros

Doble clic para editar aún cuando no esté seleccionado el elemento.

Seleccionar con un clic en general no me anda y selecciono arrastrando como workaround.

Permitir editar el FPGA pin desde el mismo modal donde está el checkbox que habilita el pin.

Las cruces para borrar cables me suelen ser muy difíciles de usar y termino borrando todos y volviendo a reconectar.

No entiendo la regla de seleccionar arrastrando cuándo es que toma o no una caja dentro del cuadrado.

Sería bueno poder intercambiar fácilmente inputs de las compuertas. Por ejemplo arrastrando un cable input sobre otro.

No encontré cómo hacer reloading de bloques importados de un archivo y mi workaround fue editar el .ice a mano.

cavearr, 2024-08-31:
Muchas gracias Martín, muchas cosas de las que comentas están muy alineadas con próximos cambios que ya tengo en marcha, lo del touch la verdad ni lo había planteado pero me parece muy interesante.

Déjame verlo en detalle y te comento cada punto pero como te digo todo muy alineado, me alegra tenerte en "el barco", vamos a construir algo muy grande entre todos!

colltoaction, 2024-11-01:
Quizás se pueda quitar la dependencia en xclip, que no viene instalado por defecto (al menos en mi Fedora 40). Es muy necesaria?