FPGAwars/icestudio #769: Migrate IceStudion from --board and --top-module flags to apio.ini

zapta, 2024-09-22:
Background: When IceStudio invokes apio commands it passes the project configuration using the flags ``--board`` and ``--top-module``.  These flags has been deprecated and may be deleted in future releases of apio, and the recommended way to convey the project configuration to apio is via a platformio like file called ``apio.ini`` file that looks like this:

```
[env]
board = icezum
top-model = main
```

This issue is for migrating IceStudio from flags based configuration to ``apio.ini`` based configuration. Implementation wise,  IceStudio can maintain a stable ``apio.ini`` file with the latest project configuration or create an ``apio.ini`` from scratch on each invocation, for example by the ``apio create`` command. 

For example, with the latter approach, the command

```
apio build --board icezum --top-module main
```

can be translated to this command sequence
```
apio create --board icezum --top-module main  --sayyes
apio build
```
Most likely, this will be a local change in the function ``apioRun``.
https://github.com/FPGAwars/icestudio/blob/63c6c71a3c705f88ff289935f129ffdb34072ea3/app/scripts/services/tools.js#L209

cavearr, 2024-09-22:
thanks @zapta i'm refactoring the toolchain module of Icestudio and i'm taking in mind your improvements in apio. In the next weeks i'm releasing it and this change in your way.

I'm maintain this thread open to check when the change will be done in Icestudio.

Thanks again!

zapta, 2024-09-22:
Thanks @cavearr. 

Please be aware of the upcoming changes in apio, the command 'apio init' is going to be replaced in the next release with 'apio create' and 'apio modify'.

If you migrate IceStudio before the next apio release, it may be easier for IceStudio to create apio.ini directly as a text file rather than using apio commands. This approach is clean and supported since apio.ini is intended to be maintained and edited by users and compatible with future apio releases.

cavearr, 2024-09-22:
Yes, this is the idea, to create apio.init i'm telling you to check it!

zapta, 2024-09-22:
On a second thought, IceStudio writing directly ``apio.ini`` is preferred over using ``apio create`` because it will allow IceStudio do use down the road ``apio.ini`` vars that are not supported by ``apio create``.

zapta, 2024-09-22:
Hi @cavearr, some considerations regarding apio.ini mutations and apio make dependencies:

Currently apio doesn't track dependencies correctly. For example if the user builds, then modifies the board in apio.ini, and then upload, the upload does not rebuild for the new board.  Same with the existing IceStudio unless if you run apio clean each time the user changes the board.

This can be fixed in apio by adding apio.ini as a dependency to apio commands such as build, assuming that IceStudio will write apio.ini only when it's really changes, to avoid unnecessary rebuilds.

In any case, I am not sure how you plan to migrate IceStudio to use apio.in but whatever you do we will make apio to handle it gracefully.

zapta, 2024-09-23:
> Hi @zapta <https://github.com/zapta> ,i'm very happy to see your
> implication with apio , go ahead!!!
>
> My plan is using all your advances in the integration with Icestudio. I'm
> now decoupling the toolchain mechanics to do more flexible in general.

Thanks Carlos!

cavearr, 2024-12-21:
Finished from 0.13.1w202412210312 wip (https://downloads.icestudio.io) all new apio comands works with apio development toolchain installed.

closed 2024-12-21 by cavearr