FPGAwars/icestudio #770: IceStudio projects fails an 'apio verify' test

zapta, 2024-09-23:
This problem was observed in the apio repository but since these looks like IceStudio projects, filing it here.

This directory https://github.com/FPGAwars/apio/tree/develop/test-examples/Alhambra-II contains three IceStudio projects.  The project ``01-LEDs-buttons`` does pass ``apio verify`` but the other two ``02-jumping-LED`` and ``03-riscv-stop-watch`` fail.

Here is an example:
```
/Users/user/projects/apio_dev/repo/test-examples/Alhambra-II/02-jumping-LED $ apio verify
iverilog -B "/Users/user/.apio/packages/tools-oss-cad-suite/lib/ivl" -o hardware.out -D VCD_OUTPUT=dummy_vcd_output -D NO_ICE40_DEFAULT_ASSIGNMENTS "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/cells_sim.v" main.v 02-jumping-LED_tb.v
main.v:383: error: 'q' has already been declared in this scope.
main.v:380:      : It was declared here as a net.
main.v:657: error: 'q' has already been declared in this scope.
main.v:655:      : It was declared here as a net.
main.v:3171: error: 'q' has already been declared in this scope.
main.v:3167:      : It was declared here as a net.
scons: *** [hardware.out] Error 3
===================================================================== [ ERROR ] Took 0.11 seconds =====================================================================
```

cavearr, 2024-12-21:
New apio commands integration finished in Icestudio wips from 0.13.1w202412210312 https://downloads.icestudio.io

All works now with new commands and apio.ini requeriments.

closed 2024-12-21 by cavearr

zapta, 2024-12-21:
Excellent!