FPGAwars/icestudio #773: Program close with ceartain combinations.

BernonR, 2024-10-14:
After several tries, I found that when I insert in the board. 2 ANDs + 2 NORs the program will close automatically. here is a GIFT

[image]

cavearr, 2024-10-14:
Thanks @BernonR  ! what version are you using?

BernonR, 2024-10-14:
version 5, is the only one I found with fewer errors than others.

cavearr, 2024-10-14:
Please, attach an screenshot of this menu panel opened:

[image]

cavearr, 2024-10-20:
Hi your gif is very small but i think this is other tool or non oficial fork.

If i'm wrong please reopen the issue and attach better image or video with more resolution.

Thanks!

closed 2024-10-20 by cavearr