FPGAwars/icestudio #774: Timing information is missing in Icestudio projects.

zapta, 2024-10-20:
This is kind of strange, and it came up while workin on the new ``apio report`` command. When building ``ice40`` and ``ecp5`` regular projects (I didn't tried on ``gowin``), nextpnr generates max clock speed information, which is important for validating the design. However, for icestudio projects for the same boards, nextpnr doesn't.   

This issue is to make icestudio projects also generating timing information.

To reproduce, run the following command in the respective apio project dir and observe if the timing info exists.

```
$ apio build --verbose-pnr | grep -i mhz
Max frequency for clock 'vclk$SB_IO_IN_$glb_clk': 127.44 MHz (PASS at 12.00 MHz)
Max frequency for clock 'vclk$SB_IO_IN_$glb_clk': 125.36 MHz (PASS at 12.00 MHz)
```

Regular projects (timing works)
https://github.com/FPGAwars/apio/tree/develop/test-examples/Alhambra-II/02-jumping-LED
https://github.com/FPGAwars/apio/tree/develop/test-examples/ColorLight-5A-75B-V8/Blinky

Icestudio projects (timing doesn't work)
https://github.com/FPGAwars/apio/tree/develop/test-examples/TB/Alhambra-II/icestudio/ledon
https://github.com/FPGAwars/apio/tree/develop/test-examples/TB/ColorLight-5A-75B-V8/icestudio/button

NOTE: The ``apio report`` command takes the timing information from a different sources (using the nextnpr ``--report`` flag) but it has the same timing issue as the ``apio build --verbose-npr``.

zapta, 2024-10-20:
Is it possible that these two icestudio projects don't use clock at all, and thus no max speed info?

cavearr, 2024-10-20:
No, it is important that the board clock be adjusted since there are blocks that depend on the calculation of that clock (although the circuit could go higher). give me a couple of days to check these projects and review what's going on.

zapta, 2024-10-20:
I found this icestudio example project that uses a clock and it does generate timing information.

https://github.com/FPGAwars/apio/tree/develop/test-examples/TB/Alhambra-II/icestudio/tff

cavearr, 2024-10-20:
Hi @zapta ! i'm checking it but i don't know i don't understand you.

This two examples has no maxspeed info because this has no timing paths in it:

https://github.com/FPGAwars/apio/tree/develop/test-examples/TB/Alhambra-II/icestudio/ledon
https://github.com/FPGAwars/apio/tree/develop/test-examples/TB/ColorLight-5A-75B-V8/icestudio/button

Please explain me what are you looking for

zapta, 2024-10-20:
@cavearr , I think that you are right,  they don't use clock and therefore the max clock speed report is not applicable to them.

I think we are good and can close this issue.  Thanks.

closed 2025-01-08 by cavearr