FPGAwars/icestudio #775: Switch icestudio to use 'apio lint' instead of 'apio verify'.

zapta, 2024-10-20:
Apio has two commands to verify the code, 'apio lint' and 'apio verify' and this is confusing. I am looking into consolidating both of them into a single 'apio lint',  possibly with a flag to achieve (reduced?) functionality similar to 'apio verify'.

This issue is for migrating icestudio to use 'apio lint' instead of 'apio verify'.  If 'apio lint' do not meet some must-have requirements, please let me know and I will try to adapt it.

cavearr, 2024-10-20:
Please, don't remove 'apio verify' until i try your new lint mode, maintain both during transition and if you want check as "deprecated" with some text in the output. 

We maintain this thread to talk about the tests

zapta, 2024-10-20:
Sounds good. Will wait for OK from you regarding the deletion of 'apio verify'.

What about 'apio time', does icestudio use it?  The new command 'apio report' superceeds it because it also provides utilization information, and timing info for ECP5 and GOWIN  ('apio time' is implemented only for ice40).

zapta, 2024-10-20:
... ``apio lint`` creates on the fly a restriction file for verilator that limits some tests. For example, for ice40 it looks like the one below. If needed, we can add a flag ``apio lint --relaxed`` that disables more lint checks as a replacement for ``apio verify``.

``hardware.vlt``:
```
`verilator_config
lint_off -rule COMBDLY      -file "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/*"
lint_off -rule WIDTHEXPAND  -file "/Users/user/.apio/packages/tools-oss-cad-suite/share/yosys/ice40/*"
```

cavearr, 2024-10-20:
i'll try 'apio time' and view how to integrate it in icestudio i think could be very useful.

zapta, 2024-10-21:
I suggest not to release icestudio with a dependency on 'apio time', It's going to be replaced by 'apio report' which is more informative and works also for ecp5 and gowin.   'apio report' is already in the apio dev branch.

cavearr, 2024-12-21:
@zapta  integration finished, Icestudio wips from 0.13.1w202412210312 with apio develpment toolchain installed works with new apio commands.

Thanks for the great work team!

closed 2024-12-21 by cavearr

zapta, 2024-12-21:
Very good. Thanks Carlos!