FPGAwars/icestudio #777: apio.ini for ECP5

tconsult, 2024-10-31:
Info:['-b''--board'] is deprecated, 
Info:['-t',--top-module'] isdeprcated. Info; Project has no apio.ini File

tconsult, 2024-11-02:
Hi cavearr  , i use Icestudio-012-win64.exe
and i could not find the apio.ini File

cavearr, 2024-11-02:
Please could you attach an screenshot of this panel menu :

[image]

tconsult, 2024-11-03:
Enclosed is a PDF File with the screenshots.

 

Thank you for your support

 

Captura.de.pantalla.2024-11-03.a.las.0.15.26.png (view on web) <https://github.com/user-attachments/assets/332fa790-1c95-4e15-aa65-e1a623e3b536>

cavearr, 2024-11-03:
Hi @tconsult ! i can't view any pdf!

tconsult, 2024-11-03:
Sorry , here as *.png Files

cavearr, 2024-11-03:
sorry @tconsult but i can't view your screenshot!

Xorlent, 2024-11-24:
@tconsult I suspect you have not completed the Toolchain setup.

1. Unplug your ECP5 from USB
2. Open Tools -> Toolchain -> Update (Latest stable)
3. After the update, click the popup dialog in the lower-right corner of the program window ("Click here to setup the drivers")
4. Reconnect your ECP5 board
5. Wait a few moments
6. Click "OK" and replace the ECP5 Development Board (Interface 0) entry with the libusbK driver
7. Once the install is done, you can close the window and unplug your board.
8. Reconnect your board and now upload your project

I hope this helps.

cavearr, 2024-12-21:
Hi! from the last wips https://downloads.icestudio.io (0.13.1w202412210312) and newer, install apio develpment from toolchain menu and all works now.

I'm closing the issue, reopen if needed.

closed 2024-12-21 by cavearr