FPGAwars/icestudio #778: Caja de código + scroll

Democrito, 2024-11-09:
Cuando se abre un circuito y hay una caja de código, sólo sale la primera línea, el resto no aparece hasta cambiar el zoom o cuando se maximiza a pantalla completa, la cosa es que cuando la ventana cambia es cuando aparece el código correctamente, con su scroll y todo eso. 

Según he visto, al menos ahora (al tratar de comprobar lo que estoy diciendo) sólo sucede si la caja de código está en el top, al navegar en módulos siempre sale bien o eso me ha parecido.

cavearr, 2024-11-10:
Desde esta wip debería estar resuelto: 0.12.2w202411100611

Democrito, 2024-11-10:
En mi Windows sigue dando guerra ese tema. Probada la versión terminada en 611

Democrito, 2024-11-11:
Vale, con esta nueva wip ya se ha resuelto. Gracias!

closed 2024-11-11 by Democrito