FPGAwars/icestudio #782: Árbol de directorios dentro de un ICE y sus propiedades corrupto

Democrito, 2024-11-09:
1.) En la edición de circuitos, a veces (no siempre) en circuitos muy complejos y que contiene un módulo dentro de otro módulo y a su vez otros módulos, etc, al re-editar estos módulos (previamente se hizo copiar y clonar antes de modificarlos), al terminar de editarse y cerrar el candado, el árbol de directorio queda corrupto y no permite llegar al "top" del ICE. Cuando esto ocurre no se puede hace nada y se pierde el circuito, ya que si lo guardas estando así, se guarda perdiendo el "top". Esto sucede sólo cuando cierras el candado para guardar los cambios (es un fallo intermitente, no hay un patrón para lograr reproducir esto).

2.) Otra cosa que sucede en estos casos de editar módulos que contienen módulos en su interior (en circuitos con módulos/ramas muy diversas) es que las propiedades del módulo con el que se trabaja en ese momento (nombre del ICE, versión, descripción, autor, etc) al cerrar el candado e ir a otro módulo, se repiten esas mismas propiedades al trabajar con otro módulo, Es decir, que el esquema sale bien, pero las propiedades del ICE pertenece al último módulo con el que se trabajó (y se cerró el candado).

cavearr, 2024-11-10:
He conseguido reproducirlo, ocurre cuando avanzas, dentro de cualquier módulo a partir del top, das a ver propiedas luego vuelves atrás, abres la edición y das a propiedades, carga el bloque que anteriormente editaste.

Aquí puede venir algún tipo de corrupción.

cavearr, 2024-11-10:
Efectivaente es siempre al volver hacia atrás. Es decir desde el top en adelante siempre funciona pero si dentro de un módulo vas atrás y editas propiedas se queda guardado el anterior.

Democrito, 2024-11-10:
Los bugs nacen, crecen, se reproducen y... con el nuevo "CucalCharliva-democritus" mueren y desaparecen.

Me vino a la mente un viejo anuncio de mata-cucarachas de cuando era joven :P

commit a59cf1b9 (2024-11-10): Changes to be committed:
========================
:bug: Fixes #782 - Project info corruption
:bug: Fixes #782 - Navigation produce file corruption
:sparkles: Fixes #780 - Improve in wire routing
:ambulance: Fixes #779 - Fix loading errors

Files to be committed:
======================
:pencil: app/scripts/controllers/design.js
:pencil: app/scripts/controllers/menu.js
:pencil: app/scripts/services/graph.js
:pencil: app/scripts/services/utils.js
closed 2024-11-10 by cavearr