FPGAwars/icestudio #784: Selección de componentes con el zoom maximizado

Democrito, 2024-11-09:
En circuitos un poco grandes o con muchos componentes, al intentar seleccionar con el ratón una zona (shift + manteniendo pulsado el botón izquierdo del ratón), se agregan componentes que no pertenece a esa área, normalmente los que están más arriba. Este problema desaparece al hacer un poco de zoom, entonces todo se seleccionar correctamente.

[image]

cavearr, 2024-11-09:
Resuelto en la última wip, cuando checkes cierra la issue.

Gracias!

jojo535275, 2024-11-09:
Little remark: i am still on version 0.11.3W202406110606
and this problem was already present on that version!

i will later update my icestudio version and check again!
Thanks Charli and Democrito for loocking at all this little but important problems!

cavearr, 2024-11-09:
Hi @jojo535275 !! in the last wip 0.12.2w202411090811 should be fixed.

I'm organizing with Democrito for a "bug haunting", i'll try to fix the next days ui bugs that are rounding for a long time and are anoying with the objetive of have a robust wip version.

If you have detected bugs please create issues  to have it in the radar.

closed 2024-11-10 by Democrito