FPGAwars/icestudio #785: Selección de componentes a vista de pájaro.

Democrito, 2024-11-09:
En circuitos un poco grandes o con muchos componentes, al intentar seleccionar con el ratón una zona (shift + manteniendo pulsado el botón izquierdo del ratón), se agregan componentes que no pertenece a esa área, normalmente los que están más arriba. Este problema desaparece al hacer un poco de zoom, entonces todo se seleccionar correctamente.

[image]

jojo535275, 2024-11-09:
Same Little remark: i am still on version 0.11.3W202406110606
and this problem was already present on thaht version!

and it doesn't make a difference if we:
-start the box on a top-left corner than move to bottom-rigtht direction
-or start the box on a bottom-rigtht corner than move to  top-left direction 

i will later update my icestudio version and check again!

Democrito, 2024-11-10:
Ya trabaja bien la selección de componentes en la última wip.

closed 2024-11-10 by Democrito