FPGAwars/icestudio #794: Recent files not work on all windows

cavearr, 2025-01-03:
@lmcapacho some users report me that the new feature not work in some windows.

Maybe could be for Spaces or Unicode characters un path.

Could do you review it?

lmcapacho, 2025-01-04:
Hi @cavearr! 

I tested it on Windows 10, and it works without any problems. The recent files list appears, and when I click on an item, the project opens successfully. I’ve attached an image of the test performed. 

Could you provide more details about the reports users have shared with you?

[image]

cavearr, 2025-01-04:
Yes, i'm trying it in linux and osx and worsk like a charm, but some windows users report me that doesn't work, i'll try to get more information.

Democrito, 2025-01-07:
Hello,

I use Windows 10 and the "open recent" option does not appear.

I leave two images with the WIP version that I am using and what I see from my Icestudio.

Any tests, questions, etc., I am here.

Greetings.

[image]  

[image]

jojo535275, 2025-01-08:
Hola Democrito!
I think on m'y win10 machine, thé function was also not there on the first
launch of icestudio (After thé installation). But i closes icestudio and
relaunch it, and in thaht second launch of the app, the function was there
?!?

Democrito, 2025-01-08:
jojo535275, It always comes out for me, since the day it was published.

A hug!

Democrito, 2025-01-08:
In case it was misunderstood, I meant that in my case that option never appears.

lmcapacho, 2025-01-08:
Hi @Democrito,

The "Open Recent" option only appears if you have previously opened a project. By default, it is hidden. Please try opening a project and check again. Let me know if I can help!

Democrito, 2025-01-08:
Hi Incapacho,

OK, I expected that option to always appear. When I saved projects, that option appeared.

Excuse my clumsiness.

Regards.

cavearr, 2025-01-08:
Thanks team!! i'm clossing the issue,thanks again for all of efforts!

Democrito, 2025-01-08:
Hi Incapacho,

The “Open Recent” option is very useful. I would like to give two suggestions, but please understand that they are just suggestions.

1.) The “Open Recent” option should always appear.

2.) Any “ICE” we open should be added to that recent list, without the need to modify and save.

About the latter, I think this approach is better because I open many “ICEs” just to copy modules from other circuits, without modifying them. This way everything will be more at hand.

Greetings and thank you very much for this new option.

lmcapacho, 2025-01-08:
Hi @Democrito,

Thanks for your suggestions!

1. I had this doubt. I think it's better if this option always appears in the menu. I made a PR to change this.

2. Recent projects are added to the recent list when they are opened, not when they are modified or saved. In PR #796, the option to add projects when launched via a file was updated. Did this option not work for you?

PS: It’s Lmcapacho, not Incapacho 😊

cavearr, 2025-01-08:
Merged, the new wip has recent files always vissible!

Thanks team! ❤️

Democrito, 2025-01-09:
Thanks for correcting me about your name, Imcapacho! I hadn't noticed, sorry.

The "Recently opened" list is added only if I open it with "Open...". That is, if I open an ICE by double clicking and then close it, and I do this several times, none of these files are saved in the "Recently opened" list. They are only saved in the mentioned list when done from "Open".

Regards.

cavearr, 2025-01-09:
@Democrito , try the last wip, @lmcapacho do this feature i think after you installed the "last wip".

Democrito, 2025-01-09:
Everything seems to be working fine now in the new WIP.

1.) "Open Recent" is always visible.
2.) All loaded "ICEs" appear in the list.

Thank you very much for the quick resolution!

cavearr, 2025-01-10:
Hi @lmcapacho !, sometimes when open Icestudio directly, the recent files appears blank:

[image]

But when open a file, the **Recent files** option is filled:

[image]

I can't been found a pattern, but i prefer reopen the issue to has this in the radar. This is not prioritary but when you want or could, check it.

lmcapacho, 2025-01-13:
Hi @cavearr !

I couldn't find a pattern either, but I added 'recentProjects' to the profile, as I considered it might be the problem.

Could you confirm whether the issue persists?

cavearr, 2025-01-13:
Thanks @lmcapacho!, i'm closing the issue, appears to be fixed!

cavearr, 2025-01-21:
Hi @lmcapacho ! time to time this bug appears, i'm reopen it without priority but put on your radar!

lmcapacho, 2025-01-21:
Hi @cavearr !

You're right, this bug appears for me too. I checked it and found that the profile doesn't load in time when reading the recent projects list. I forced the profile to load to solve this issue.

closed 2025-01-21 by cavearr

cavearr, 2025-01-21:
I just tried it and it seems to work fine!!!

One thing @lmcapacho, I suggest these tasks to you because you did the first PR  on this feature and I have many other tasks going on and I prefer that if you want and feel like it, you can fix it yourself. But every time I refer to you, if you don't have time or don't want to do it, please tell me with confidence and I'll put it on the list.

I don't want you to take any of this as an imposition.

Today I will finish the spec for changing the UI in the forms. I will send it to you in a while.

lmcapacho, 2025-01-22:
Hi @cavearr ,

No worries at all! I really enjoy working on this project—it’s amazing. As a user and a fan of free software, I’m happy to help wherever I can.

PR #807 (2025-01-21): Force load profile to fetch recent projects
Closes #794