FPGAwars/icestudio #806: Button style in close window dialog

cavearr, 2025-01-21:
@lmcapacho thanks for this feature, i change the literals to do more clear, if you think are few large, try other options but it is important define clearly the operative (save and close, etc).

If you could put a "red ribbon" to this develop, define a better style to the buttons, according to other ui buttons of iceStudio:

[image]

Thanks!

lmcapacho, 2025-01-21:
Hi @cavearr 

Thanks for your feedback!

Previously, the buttons were named "**Close**" and "**Cancel**". How do feel about changing them to "**Save and Close**", "**Discard and Close**", and "**Cancel**"?

As for the button styles, they currently match the design used in other windows and buttons in Icestudio, following the theme used in Alertify. Could you share more about your idea?

cavearr, 2025-01-21:
as you see in white theme (i don't try yet in dark theme) there are no buttons, appears links. i think we need to define a thin border like the "Import" in verilog form.

TimRudy, 2025-01-21:
> @lmcapacho <https://github.com/lmcapacho> For me, my thoughts:
>
>    1. 💾 Save & Quit
>    2. NO save & Quit
>    3. Back ⏎
>
> How do you think? Cancel is ambiguous because what do you cancel, the
> dialog, the save , .....i think back is clearly, the button return to your
> previous state.
>
> With "quit" we gain small word and clear action.
>
> For save changes, i think we should use save for both cases, because
> "cancel" is again ambiguos , what do you want to cancel? (the dialog, the
> changes, the windows closing....).
>
> We could use unicode icons like save or back to do emphasis in the action.
> the point 2 i don't find an standard icon that like me, but i think could
> be fine without icon.

I think it is common to say "Save and Quit" plus "Quit without saving" or
"Quit without save"  . [The capitalization of the words, i.e.
"saving"/"Saving"/"Save" is a question. We can search for
commonly used pattern or ask ChatGPT.]

lmcapacho, 2025-01-22:
I was exploring patterns in some open source software and found the following examples:

- LibreOffice: Save, Don't Save, and Cancel
- Arduino IDE: Save, Don't Save, and Cancel
- Gnome Text Editor: Save, Discard, and Cancel
- Audacity: Yes, No, and Cancel
- GIMP: Discard Changes and Cancel

In other platforms, like Vivado, they also use: Save, Don't Save, and Cancel.

It seems that these button names follow this structure because the question being asked is so clear. What do you think?

cavearr, 2025-01-23:
Good research!! thanks @lmcapacho and @TimRudy !

I vote for Save, Don't save and Cancel

cavearr, 2025-01-23:
Closed in the #808 PR

cavearr, 2025-01-23:
Thanks team!

closed 2025-01-23 by cavearr