FPGAwars/icestudio #876: Automatic COMMITs and PRs checkers and PRE-COMMIT script for local development.

cavearr, 2025-02-16:
@TimRudy report some bugs in the pre-commit flow, check message at discussion:

### Discussed in https://github.com/FPGAwars/icestudio/discussions/875

cavearr, 2025-02-16:
@TimRudy the fix is done, you need to pull and execute **npm install --legacy-peer-deps** or copy **scripts/git/pre-comit** to .git/hooks/pre-commit

Tell me if all works .

About the point 2) are you referring to this?:

[image]

Are you building on Windows? i'm never build the icestudio package on windows, i always build from linux. if this is the problem, i'll try to research how fix it to build on windows. If is it in other place tell me please to clarify the problem.

Thanks again!

TimRudy, 2025-02-16:
The above `mkdir -p`: I removed it and added quotes to make it work, Jan 30:
before: `mkdir -p ${DIST_ICESTUDIO_WIN64}/resources/images`
after: `mkdir "${DIST_ICESTUDIO_WIN64}/resources/images"`

Feb 2, you did `"RUN_BUILD --> Fix wrong mkdir command in Windows OS"`

Having said this, on Git Bash I test the cmd that's being run today in pre-commit:
`$ mkdir -p "$TEMP_DIR"`
on my Git Bash and it works perfectly! So the error running "winico" is another "run-time"/environment problem

cavearr, 2025-02-16:
i understan better now!, yes i added again in winico because gruntfile not segmented well the tasks and when build the package breaks because in linux if you don't use -p in the command, the folders are not created recursively.

For example if you put:  mkdir /dir1/dir2/dir3 and not exists any of them, the command fails, but if you add the -p argument , mkdir create all of them.

Gruntfile uses node and node "hide" the OS with a compatibility layer, i'll try to reproduce the break in winico because could be some other bug as you tell.

Thanks again @TimRudy for the detailed feedback!

cavearr, 2025-02-16:
If you could give me a sequence of actions to replicate the error with mkdir on windows, could be very useful.

TimRudy, 2025-02-17:
Apologies, that error was when I ran `npm run buildWindows` in PowerShell, not in Git Bash terminal.
I won't do that any more!

TimRudy, 2025-02-17:
Now,

=> 3. There's a problem when I hit nwjs build (using Git Bash). After it does the 'copy' and 'json-minify' successfully it errors at:
```
Running "nwjs:src" (nwjs) task
Error: Either the specific version info does not exist or the version manifest itself does not exist. In case of the latter, please check your internet connection...
```
I don't know if you can reproduce that.
Checking in the call stack the problem is at
```
...\icestudio\node_modules\nw-builder\src\index.js:

await nwbuild(options);
  -> await util.validate(options, releaseInfo);
```
Here `releaseInfo` is undefined.
`options` I will list only some of the fields here:
```
options = {
  app: {...},
  arch: 'x64',
  downloadUrl: 'https://dl.nwjs.io',
  flavor: 'sdk',
  managedManifest: false,
  manifestUrl: 'https://nwjs.io/versions',
  platform: 'win',
  srcDir: './dist/tmp',
  version: '0.95.0-sdk',
  ...
}
```
It has successfully downloaded a `manifest.json`, it seems, so I see the file at:
```
...\icestudio\cache\manifest.json
```

The node version that is being used is: `v20.12.2`.
In `manifest.json` I notice there is no `20.12`. That is probably the problem. What does it mean however...?
I also saw the nwjs `flavor: 'sdk'` and the `'0.95.0-sdk'`. Not a problem with that...?

TimRudy, 2025-02-17:
P.S. This is a change I have for a PR because it installed the hook but then it stopped and did not do any more!

[image]

cavearr, 2025-02-17:
You are right @TimRudy !! i we update the pre-commit , never updates than the first!

Do you want to  do i? the idea will be remove the check for if pre-commt exists (sh -c...) and give that always npm  and grunt tasks works.

The idea will be in setupDevEnv check i f pre-commit file exists in git hooks and if exists check with for example md5 if is the same that scripts file, if is different replace.

I know this check could not be needed but if setupDevEnv grow in checks  could be important.

About the 0.95.0-sdk i need to add to windows build a sed command, this is in github actions but if you run manually breaks, for build npm needs that the version of nwjs be 0.95.0 without sdk, this is on my side to check that works on local and in github.

If you want, you could  create the issues ok? i'm try to fix this today.

Thanks again @TimRudy !

cavearr, 2025-02-17:
hi @TimRudy **start** task is updated and now always the pre-commit is overwritten to assure the update if is needed.

I'm checking the way to build in windows (i only build packages in linux and osx, i'm not sure if is possible for the moment build windows because all the tools are linux cross tools for windows. maybe with wsl i'm research arount it.

TimRudy, 2025-02-17:
I'm testing the scripts on Windows and it works pretty well. Thanks for the tip about "sed" for the "-sdk" because now `buildWindows` gets further.

However, we are back to point (2).

I run this command on Git Bash and works great:
```
$ mkdir -p "./dist/icestudio/win64/resources/images" && cp docs/resources/images/logo/icestudio-logo.ico ./dist/icestudio/win64/resources/images
```
But in the context of the grunt task, something is different about the PATH maybe and it reverts to calling the native Windows mkdir:
```
Running "shell:winico" (shell) task
A subdirectory or file -p already exists.
Error occurred while processing: -p.
Warning: Command failed: mkdir -p "./dist/icestudio/win64/resources/images" && cp docs/resources/images/logo/icestudio-logo.ico ./dist/icestudio/win64/resources/images
A subdirectory or file -p already exists.
Error occurred while processing: -p.
```

I think that could be overcome by injecting the correct `/usr/bin/mkdir.exe` (below) at top of PATH inside grunt task.

Near the top of my path is `mkdir.exe` for bash and Linux and it is the good one:
`$PATH = /c/Users/tim/bin:/mingw64/bin:/usr/local/bin:/usr/bin:...` =>
```
$ /usr/bin/mkdir -p "./dist/icestudio/win64/resources/images" && cp docs/resources/images/logo/icestudio-logo.ico ./dist/icestudio/win64/resources/images
```

My problem right now is I don't know how/where the native windows `mkdir` is being called.
Even if I go try extreme case, on PowerShell:
`> cmd /c which mkdir`
it lists the good one:
`/usr/bin/mkdir`.

Or on Git Bash, if I test "Git" additions to the top of path, they do not cause a problem, i.e. this works:
`$ "C:/Program Files/Git/usr/bin/mkdir" -p ./dist/icestudio/win64/resources/images && cp docs/resources/images/logo/icestudio-logo.ico ./dist/icestudio/win64/resources/images`

TimRudy, 2025-02-17:
Mentioning that the scripts all run on Windows.
- the above "mkdir -p" does not fail if there is no `icestudio/-p`. Then when there is one, it fails :-)

- buildWindows completes: the installer works and it installs the working Icestudio!

[image]

cavearr, 2025-02-17:
Wohoo you are the best!!! 🤟

cavearr, 2025-02-18:
i'm closing this issue if you need reopen it.

TimRudy, 2025-02-18:
Well, I confused the conversation. There is still point (2), "winico" could use a fix:

In context of the grunt task, the native Windows `mkdir` is called, instead of correct Linux/bash `mkdir` which is `/usr/bin/mkdir.exe`.

Something may be different about the PATH in that context. Why doesn't bash `mkdir` have precedence? For Windows, `mkdir -p` is not acceptable. The `-p` is not needed to create recursively (as it is in Linux). Instead it creates a directory `-p`. This error will only be noticed the second time:
```
Running "shell:winico" (shell) task
A subdirectory or file -p already exists.
Error occurred while processing: -p.
Warning: Command failed: mkdir -p "./dist/icestudio/win64/resources/images" && cp docs/resources/images...
A subdirectory or file -p already exists.
Error occurred while processing: -p.
```

- This could be overcome by injecting the correct `/usr/bin` as higher precedence in PATH inside grunt task.
- Or there could be a separate declaration of `winico command` for Win platform that just says:
`mkdir "${DIST_ICESTUDIO_WIN64}/resources/images"`
but this is not a good solution, because everything on Win is supposed to run under Git Bash and act mostly just as Linux; and it's an incorrect command if it really did execute under Git Bash!
- A hack solution would be to automatically delete `-p` directory if it happens to be created!

cavearr, 2025-02-18:
i'm reopening it to do more tests!

cavearr, 2025-02-23:
I'm doing for closest

closed 2025-02-23 by cavearr