FPGAwars/icestudio #878: Use "-sdk" removal filter for "nw": "0.95.0-sdk" retrieved from package.json, for "buildWindows" locally (because no GitHub Action)

TimRudy, 2025-02-17:
For running `buildWindows` locally via `npm`, a filter is needed like this in `.github\workflows\build-windows-manual.yml`:
`sed -i 's/\("nw": "[^"]*\)-sdk"/\1"/g' package.json`

It should pass the correct NWJS to "nwjs:src" build - i.e. must use name "0.95.0" instead of "0.95.0-sdk". Don't use the `sed` command. It changes `package.json` in-place.

Aside from this, "buildWindows" will work on Windows if:
- developer installs `gettext0.23-iconv1.17-static-64.exe` or latest version
- developer installs `NSIS.exe`
- the `"mkdir -p"` problem in Gruntfile is resolved (in another Issue)

cavearr, 2025-02-17:
Do you install manually this two packages? in the github actions i install "manually" with the operating system in fact. Maybe we coud improe the build scripts to detect os and install this dependencies automatically, i'm sure this improve the developer experience.

TimRudy, 2025-02-18:
Yes I installed those when I found they were needed.

If I'm thinking about this, it is not too important that a developer can use all the scripts, like "buildWindows" or the others that create an installer. Usually, it's just the pipeline that handles these build-release items.

Why don't we just document these so that any developer knows it's possible? - even though they're probably not going to do it...
1. Install gettext-iconv to do `preptext` or - actually - anything like `npm start`...
2. Install NSIS.exe to do `buildWindows`
3. Run `sed -i 's/\("nw": "[^"]*\)-sdk"/\1"/g' package.json` to temporarily modify your `package.json` to do `buildWindows`
4. Make sure `"-p"` directory is cleaned-up/absent because of a "known issue" if you do `buildWindows`

Like that? The hard work of builds & automation is already done. Tell me a place to document these and we can be finished with the issue.

cavearr, 2025-02-23:
Hi @TimRudy! I would like icestudio to have its own development guide and minimal user guide, without wikis or external pages.

For this, I put a new directory inside /docs called developers.

If you agree with this, I think we could create Markdown files in it with the different aspects of the development to make the work easier in the future. I'm trying different Markdown parsers to generate after that "a page" that could be easily opened from Icestudio or in a browser (on github I could see the Markdown renderer, but it would be better if we could open it inside Icestudio).

With the user manual I think we will do the same, I need to finish a couple of things that will change the user interface, and once finished I want to make some quick guides that are within Icestudio itself. Then we can put references to the wiki and other documentation sites.

Thanks!