FPGAwars/icestudio #886: bug: Toolchain installation fails under Windows

ManfredSchmitz, 2025-03-27:
### Steps To Reproduce

Install Icestudio (stable version as well as 0.13.4w202502260302)
Run Icestudio
-> Toolchain not installed. Click to install
-> Apio install driveres fails after 80%

### Expected behavior

[Icestudio fails.pdf](https://github.com/user-attachments/files/19490016/Icestudio.fails.pdf)

[icestudio.log](https://github.com/user-attachments/files/19490049/icestudio.log)

### Actual behavior

Icestudio is not usable - unfortunately - it looks very promissing

### Web console trace

```console

```

cavearr, 2025-03-27:
Thanks for your feedback, i'll try to reproduce your problem appears a conflict with something in your installation, i'm telling you soon.

cavearr, 2025-03-27:
Hi! could you tell me if this path exists on your computer and if exists  if is exactly with lower an uppercase? C:\SPB_DATA\.icestudio\venv\Scripts\

ManfredSchmitz, 2025-03-28:
Hi,

That is the exact path:

C:\SPB_Data\.icestudio\venv\Scripts

.icstudio is a subdirectory. There is a leading "\"

Regards,
Manfred

________________________________
Von: Carlos Venegas Arrabé
Gesendet: Donnerstag, 27. März 2025 22:36
Bis: FPGAwars/icestudio
Cc: Manfred Schmitz; Author
Betreff: Re: [FPGAwars/icestudio] bug: Toolchain installation fails under Windows (Issue #886)

[cavearr]cavearr left a comment (FPGAwars/icestudio#886)<https://github.com/FPGAwars/icestudio/issues/886#issuecomment-2759528430>

Hi! could you tell me if this path exists on your computer and if exists if is exactly with lower an uppercase? C:\SPB_DATA.icestudio\venv\Scripts\

cavearr, 2025-03-29:
Hi , could be possible that this path in your PATH "C:\Users\ManfredSchmitz\AppData\Local\Programs\Microsoft VS Code\bin" breaks  apio (that is other FPGAwars software that runs under Icestudio and this not support spaces in the path for a Yosys limitation.
Could you uninstall Microsoft VS Code and reinstall it without spaces in the directory, check reboot your computer for PATH refresh and try it again?

efa, 2025-07-28:
take care that OrCAD redefine the HOME env var also if it already defined in your OS, pointing it to C:\SPB_Data