FPGAwars/icestudio #905: bug: Error executing command set APIO_HOME_DIR="C:\Users ...

efa, 2025-07-25:
### Steps To Reproduce

Steps to reproduce behavior
    Win10
    Download and install Python 3.13.x
    Download and install icestudio for windows (stable): icestudio-0.12-win64
    Launch icestudio
    Select board: Alhambra-II
    Click message that prompts to install STABLE Toolchain
    A pop-up appears with a single OK button: "Install the STABLE Toolchain. This operation requires Internet connection. NOTE: You need to disconnect your VPN (if any) to allow the toolchain installation Do you want to continue?"
    Click OK -> start installation of the toolchain, it end with error below:
bug: Error executing command set APIO_HOME_DIR="C:\Users ...
see attache image below:
[image]

### Expected behavior

no error on install toolchain

### Actual behavior

Error executing command set APIO_HOME_DIR="C:\Users ...

### Desktop (please complete the following information)

Win10
Firefox 141.0

### Web console trace

```console

```

efa, 2025-07-28:
I repeated the test with Alhambra-II connected, same result

efa, 2025-07-28:
I tryed to guess the path as in the dialog error are cut.

I just manually add these set to environment variable (user env, not system env):
```
APIO_HOME_DIR=C:\Users\<user>\.icestudio\apio
PATH=<prevPATH>;C:\Users\<user>\.icestudio\venv\Scripts
```
then remove the toolchain and reinstall, same result

cavearr, 2025-07-28:
I'm working hard these days to finalize a new version of IceStudio with tons of improvements. This afternoon, I'll release a new WIP version. Although it's still in progress, it has many improvements over the stable version. I'll let you know when it's ready in case you want to try both versions. Regarding what's happening with Windows, it sounds like a driver issue. I'll try to find a way to help.

efa, 2025-07-28:
are you confident that Debian version work?

cavearr, 2025-07-28:
stable version is not compatbile now with the latest toolchain , in next weeks, we realease a new version with all the support.

Thanks and sorry for the apologize , now are very changing moment for icestudio.

closed 2025-07-28 by cavearr

efa, 2025-07-28:
don't worry, my main PC is Debian based, only the company PC is Windows, and I'm doing this for hobby so, I very satisfied and hope test/debug are useful to you